chipsalliance/caliptra-sw #100: ROM Task - support measurement command

varuns-nvidia, 2023-03-23:
Problem statement: SoCs may have ROM patches and firmware that they need to boot before Caliptra firmware. Want to submit those measurements to Caliptra before execution.

 

Solution:
- Caliptra ROM will add a mailbox command for storing measurement.
- Caliptra ROM will support this command prior to CALIPTRA_FW_LOAD command.
- Caliptra ROM will use these measurements to derive CDI.
- Caliptra ROM will store those measurements to either PCRs or to Data Vault (set of write lockable registers on pwrgood reset). Vishal prefers Data Vault because PCRs don't support write locking, but write locking is not a strict requirement.

varuns-nvidia, 2023-03-28:
Amending the ask: ROM patch measurements should NOT be used as part of CDI derivation for Alias FMC.

jhand2, 2023-03-28:
> Caliptra ROM will use these measurements to derive CDI

So the new proposal instead is that runtime will push these measurements into DPE during initialization, right?

FerralCoder, 2023-03-28:
Great, I like this proposal much better!

varuns-nvidia, 2023-04-25:
Discussion with @vsonims  and @mhatrevi :
* ROM to implement https://github.com/chipsalliance/caliptra-sw/tree/main/runtime#stash_measurement in a command loop
* Handle 8 measurements prior to CALIPTRA_FW_LOAD
* Caliptra ROM is producing a fixed size event log in DCCM. ROM will populate this event log with the metadata and measurement arguments provided by the STASH_MEASUREMENT calls. Later, Runtime FW can use this event log to build DPE TCIs
* Caliptra ROM to extend PCR[1] with the measurement arguments provided (NOT the metadata)
* @vsonims to discuss with @bharatpillilli whether ROM should perform LDevID key generation before or after the STASH_MEASUREMENT command loop. Current stance is to do it before (keep existing flow, unify command loops with CALIPTRA_FW_LOAD)

@jhand2 @FerralCoder does this look good?

bharatpillilli, 2023-04-26:
My suggestion/view is thatbefore we accept any payload from outside of Caliptra, we should complete the LDevID key gen operation.

mhatrevi, 2023-04-28:
1. ROM will implement the following command:
[image]

2. ROM will extend the measurement field of the command into PCR1.
3. ROM will allow extending up to 8 measurements.
4. These command(s) will be handled before handling the CALIPTRA_FW_LOAD command to download the firmware image.
5. These commands won’t be handled once the CALIPTRA_FW_LOAD command is received.
6. A PCR log will be maintained in the DCCM for all the PCR extensions.
7. PCR log entry format will be as follows:
        •	ID (1 Byte)
        •	PCR ID (1 Byte): PCR to which the data is being extended to.
        •	Data (up to 48 bytes)
        •	Metadata (4 bytes)

bluegate010, 2023-04-28:
* ROM should guarantee that this data is written only during cold boot and not modified during subsequent warm resets.
* ROM should support a command to disable attestation until next cold boot. Use-case: all eight measurements have been provided, and then firmware load fails, and SoC does not want to halt boot.

varuns-nvidia, 2023-04-28:
@bluegate010 : should ROM do DISABLE_ATTESTATION automatically if it hits the max count?

bluegate010, 2023-04-28:
Good question. My gut initial feeling is to leave it to the SoC how to deal with that error condition, that nuking the DICE identity is an awful big hammer to put on a tripwire like that. However, on reflection the only other thing I can think of for an SoC to do in that situation is to just halt boot. So, may as well just do the safe thing implicitly.

varuns-nvidia, 2023-06-28:
@mhatrevi what's the status of this issue?

mhatrevi, 2023-06-30:
@varuns-nvidia This is still being discussed internally. Hoping to get clarity soon.

mhatrevi, 2023-09-12:
https://github.com/chipsalliance/caliptra-sw/pull/559

closed 2023-09-12 by mhatrevi