chipsalliance/caliptra-sw #108: hmac384::test_hmac3 fails with latest caliptra-rtl

korran, 2023-03-28:
With the old RTL, no problem...
```
$ git checkout main
$ (cd hw-latest/caliptra-rtl/ && git log -2 --oneline)
6b69da9 (HEAD) Fix AHB multiplexing issue: #20
01fcfd1 Initial Pre0p8 release from dev-msft (#9)
$ export CXX="ccache g++"
$ cargo test --release --features=verilator -p caliptra-lib test_hmac384
<snip>
hmac384::test_kat...    [ok]
hmac384::test_hmac0...  [ok]
hmac384::test_hmac1...  [ok]
test test_hmac384 has been running for over 60 seconds
hmac384::test_hmac3...  [ok]
<snip>
```

With the new RTL...
```
$ git checkout rtl-2023-03-27
$ (cd hw-latest/caliptra-rtl/ && git log -8 --oneline)
cdd9afa (HEAD, ahb-mux-fix) Fix AHB multiplexing issue: #20
fa91d66 (main) Merge pull request #27 from chipsalliance/dev-integrate
f47d19d Sync from MSFT Internal repo w/ TRNG, HMAC SCA, PCR Hash/sign, Env Var rename (#21)
e5c1b48 Checking in a copy of the opentitan repository taken at (#22)
3678553 Adding verilator CI smoke test (#19)
b19f4ff Add Register docs links to README.md (#13)
01fcfd1 Initial Pre0p8 release from dev-msft (#9)
$ cargo test --release --features=verilator -p caliptra-lib test_hmac384
hmac384::test_kat...    [ok]
hmac384::test_hmac0...  [ok]
hmac384::test_hmac1...  [ok]
test test_hmac384 has been running for over 60 seconds
hmac384::test_hmac3...  [failed]
Error: panicked at 'assertion failed: `(left == right)`
  left: `Array4xN([1097496930, 177872290, 3861325564, 3222974378, 3615216998, 3373329619, 1992284781, 2272737848, 311909870, 2326523809, 2038415403, 775445936])`,
 right: `Array4xN([3806224643, 3931861133, 1050772493, 1774245656, 4173031246, 1110546316, 4049204051, 1657138985, 2965254665, 1955743625, 2383417612, 2640588215])`', drivers/test-fw/src/bin/hmac384_tests.rs:122:5

* TESTCASE FAILED
```

The test case does the following:

```
fn test_hmac3() {
    //
    // Step 1: Place a key in the key-vault.
    //
    let seed = [0u8; 48];
    let mut key_usage = KeyUsage::default();
    key_usage.set_hmac_key(true);
    let key_out_1 = KeyWriteArgs {
        id: KeyId::KeyId0,
        usage: key_usage, // hmac_key
    };
    let result = Ecc384::default().key_pair(
        Ecc384Seed::from(&Ecc384Scalar::from(seed)),
        Ecc384PrivKeyOut::from(key_out_1),
    );
    assert!(result.is_ok());

    //
    // Step 2: Hash the data with the key from key-vault.
    // Key is [0xc9, 0x8, 0x58, 0x5a, 0x48, 0x6c, 0x3b, 0x3d, 0x8b, 0xbe, 0x50, 0xeb, 0x7d, 0x2e, 0xb8, 0xa0,
    //         0x3a, 0xa0, 0x4e, 0x3d, 0x8b, 0xde, 0x2c, 0x31, 0xa8, 0xa2, 0xa1, 0xe3, 0x34, 0x9d, 0xc2, 0x1c,
    //         0xbb, 0xe6, 0xc9, 0xa, 0xe2, 0xf7, 0x49, 0x12, 0x88, 0x84, 0xb6, 0x22, 0xbb, 0x72, 0xb4, 0xc5,];
    //
    let data: [u8; 8] = [0x48, 0x69, 0x20, 0x54, 0x68, 0x65, 0x72, 0x65];

    let result: [u8; 48] = [
        0xe2, 0xde, 0x61, 0x3, 0xea, 0x5b, 0x70, 0x8d, 0x3e, 0xa1, 0x84, 0xd, 0x69, 0xc0, 0xd7,
        0x18, 0xf8, 0xbb, 0x67, 0x4e, 0x42, 0x31, 0x97, 0x8c, 0xf1, 0x59, 0xf3, 0x53, 0x62, 0xc5,
        0xef, 0x29, 0xb0, 0xbe, 0x32, 0x9, 0x74, 0x92, 0x47, 0x89, 0x8e, 0x10, 0x11, 0xc, 0x9d,
        0x64, 0x2d, 0xb7,
    ];

    let mut out_tag = Array4x12::default();
    let key = KeyReadArgs::new(KeyId::KeyId0);
    let actual = Hmac384::default().hmac(key.into(), (&data).into(), (&mut out_tag).into());

    assert!(actual.is_ok());
    assert_eq!(out_tag, Array4x12::from(result));
}
```

It's not clear to me whether the test-case or the RTL is faulty.

bharatpillilli, 2023-03-28:
@mojtaba-bisheh - can you check if this is due to the countermeasure code that went in? If so, it could be that HMAC API is not updated on FW  side

mojtaba-bisheh, 2023-03-28:
The countermeasure itself cannot cause failure in hmac since it doesn't change the output. I'm working with @mhatrevi to find out what's the issue here.

mhatrevi, 2023-03-28:
Issue is that the new nonce functionality is not implemented for the ECC384 driver. The failing test case (test_hmac3) is using the ECC384 driver to generate an HMAC key (via key_pair generation) and thus the key generated is different, causing the test the to fail. I have the nonce changes ready and tested, they are awaiting review.

korran, 2023-03-29:
Can confirm that @mhatrevi's commit fixes the test:

```
$ git log -1 --oneline
75886e1 (HEAD -> rtl-2023-03-27, chips/rtl-2023-03-27) [feat] Adding nonce input argument to Ecc384 firmware api
$ (cd hw-latest/caliptra-rtl/ && git log -1 --oneline)
cdd9afa (HEAD, origin/ahb-mux-fix, ahb-mux-fix) Fix AHB multiplexing issue: #20
$ cargo test --release --features=verilator -p caliptra-drivers test_hmac384
<snip>
hmac384::test_kat...    [ok]
hmac384::test_hmac0...  [ok]
hmac384::test_hmac1...  [ok]
test test_hmac384 has been running for over 60 seconds
hmac384::test_hmac3...  [ok]
<snip>
```

commit 75886e1d (2023-03-29): [feat] Adding nonce input argument to Ecc384 firmware api

This fixes #108. This change adds the capability to the Ecc384 firmware
api to ingest a nonce.  This corresponds to the similar functionality
added to the  ECC384 RTL crypto block.
commit 01bc419b (2023-03-29): [feat] Adding nonce input argument to Ecc384 firmware api

This fixes #108. This change adds the capability to the Ecc384 firmware
api to ingest a nonce.  This corresponds to the similar functionality
added to the  ECC384 RTL crypto block.
closed 2023-03-29 by korran