chipsalliance/caliptra-sw #128: Central error code registry for Caliptra

rusty1968, 2023-03-31:
Diagnostic and error reporting capability should reside in a common crate with all the caliptra error codes, that can be referenced from test cases and caliptra clients, including a Debug impl when compiled with the std feature.

rusty1968, 2023-04-28:
/// The purpose the error handling module is to provide a common error handling
/// and reporting mechanism for all Caliptra components.
/// Caliptra components are entities that can generate errors. Examples of
/// Caliptra components are the Driver, ROM, FMC, RT, and FIPS.
/// Each Caliptra component has a unique identifier that is used to identify
/// the component when reporting errors.
///
/// The error handling module shall reside in the common crate of the Caliptra
/// project. The error handling module shall be used by all Caliptra components.
///
/// Error handling and reporting is done by raising an error. Caliptra uses the
/// idiomatic Rust approach of returning a Result type from functions that can
/// fail. The Result type is defined as follows:
/// CaliptraResult<T> = Result<T, CaliptraError>
///
/// Caliptra error policy partitions the 32-bit name error space into three parts:
/// 1. Caliptra component identifier  (8 bits))  
/// 2. Caliptra sub-component identidier (8 bits)
/// 3. Caliptra error code (16 bits)
///
/// See the diagram below for the error space partitioning:
///
/// | 31-24  |  23-16  |  15-0  |
/// |--------|---------|--------|
///
/// The Caliptra sub-component identifier is Caliptra component specific.
/// The Caliptra error code is Caliptra sub-component specific.
///
/// Non fatal error flow propagates all the way to the topmost module of the Caliptra
/// component. The topmost module is the one that is responsible for reporting
/// the errors to the system in which the Caliptra .
///
/// Error reporting is done via SoC interface registers, more specifically the
/// fatal and non-fatal error register via the API defined in
/// caliptra::drivers::error reporter.
///
/// Fatal errors are errors which trigger exceptions and NMI. Fatal errors are
/// reported  via fatal error registers.
/// Non-fatal errors
/// are errors which are encountered in the course of normal operation and
/// reported to the user via the non-fatal SoC interface registers.
///
pub enum CaliptraComponent {
    DRIVER = 0x00,
    ROM = 0x01,
    FMC = 0x02,
    RT = 0x03,
    FIPS = 0x04,
}

FerralCoder, 2023-09-13:
https://github.com/chipsalliance/caliptra-sw/pull/274

closed 2023-09-13 by FerralCoder