chipsalliance/caliptra-sw #129: Mailbox driver does not detect incoming messages correctly

korran, 2023-03-31:
From https://github.com/chipsalliance/caliptra-rtl/blob/main/docs/Caliptra_Integration_Specification.pdf:

> ### 9.3 Sender Protocol
> 
> Sending data to the mailbox:
> 
> 1. Requester queries the mailbox by reading the LOCK control register.
>      - If LOCK returns 0, LOCK is granted and will be set to 1.
>      - If LOCK returns 1, MBOX is locked for another device.
> 2. Requester writes the command to the COMMAND register.
> 3. Requester writes the data length in bytes to the DLEN register.
> 4. Requester writes data packets to the MBOX DATAIN register.
> 5. Requester writes to the EXECUTE register.
> 6. Requester reads the STATUS register. Status can return:
>      - CMD_BUSY - 2’b00 – Indicates the requested command is still in progress
>      - DATA_READY - 2’b01 – Indicates the return data is in the mailbox for requested command
>      - CMD_COMPLETE- 2’b10 – Indicates the successful completion of the requested command
>      - CMD_FAILURE- 2’b11 – Indicates the requested command failed
> 7. Requester reads the response if DATA_READY was the status.
> 8. Requester resets the EXECUTE register to release the lock.
> 
> ### 9.4 Receiver Protocol
> 
> Upon receiving indication that mailbox has been populated, the appropriate device can read the
> mailbox. This is indicated by a dedicated wire that is asserted when Caliptra populates the mailbox for SoC consumption.Receiving data from the mailbox:
> 
> 1. On mailbox_data_avail assertion, the receiver reads the COMMAND register.
> 2. Receiver reads the DLEN register.
> 3. Receiver reads the MBOX DATAOUT register.
>    - Continue reading MBOX DATAOUT register until DLEN bytes are read.
> 4. If a response is required, receiver can populate the mailbox with the response by writing DATAIN
> 5. Set the mailbox status register to hand control back to the sender to read the response
> 6. The sender will reset the EXECUTE register after reading the response.
>    - This releases the LOCK on the mailbox. 

See this code in the mailbox driver:

https://github.com/chipsalliance/caliptra-sw/blob/1b4af7aaed993042891df38b4822145bc1805866/drivers/src/mailbox.rs#L67-L77

This is incorrect. Nowhere in section 9.4 do the docs mention a receiver reading the status field, which is only used to notify the original sender (the SoC) of the result. The driver should not be looking at `mbox.status().read().status()`, but instead `mbox.status().read().mbox_fsm_ps() == MboxFsmE::MboxExecuteUc`, which becomes true after the SoC sets the execute bit. (The docs say to use the interrupt, but I can't find docs stating which interrupt line comes from the mailbox, and the fsm state field seems to work well enough).

Because of this mistake, tests pretending to be the SoC are setting the data-ready bit manually, which is also incorrect:

https://github.com/chipsalliance/caliptra-sw/blob/1b4af7aaed993042891df38b4822145bc1805866/hw-model/src/lib.rs#L213-L219

https://github.com/chipsalliance/caliptra-sw/blob/1b4af7aaed993042891df38b4822145bc1805866/hw-model/src/lib.rs#L213-L219

All this has to be fixed, and the emulator needs to be modified to behave the same as the RTL.

korran, 2023-03-31:
@vsonims, @Nitsirks, and @calebofearth for visibility.

rusty1968, 2023-04-01:
@korran , this is what the Caliptra Integration Specification  stated when I implemented this : 

9.4	Receiver Protocol
Upon receiving indication that mailbox has been populated, the appropriate device can read the mailbox. This is indicated by a dedicated wire that is asserted when Caliptra populates the mailbox for SoC consumption, **also by the STATUS register returning DATA_READY**

Seems behavior was modified and we were not aware of it.

closed 2023-04-12 by korran