chipsalliance/caliptra-sw #134: Handoff table Indexes can't fully  describe storage entry address in the data vault (lossy conversion)

rusty1968, 2023-04-03:
The address of an entry stored in the data vault is a pair  (entry_type, entry_variant) , where entry type can be WarmResetEntry4, ColdResetEntry48,  The entry type is an abstraction of the register type (sticky/non-sticky/scratch)  

The entry type information is lost when conversion entry_type::entry_variant to u8 is performed forcing FMC to make assumptions about the entry type  where the information is stored and defeating the purpose of a handoff table to decouple FMC and ROM.  

Example : 
 rt_fw_load_addr_idx: WarmResetEntry4::RtLoadAddr.into(),
fmc_pub_key_x_dv_idx: ColdResetEntry48::FmcPubKeyX.into()

pub enum WarmResetEntry4 {
    RtSvn = 0,
    RtLoadAddr = 1,
    RtEntryPoint = 2,
    ManifestAddr = 3,
}
pub enum ColdResetEntry48 {
    LDevDiceSigR = 0,
    LDevDiceSigS = 1,
    LDevDicePubKeyX = 2,
    LDevDicePubKeyY = 3,
    FmcDiceSigR = 4,
    FmcDiceSigS = 5,
    FmcPubKeyX = 6,
    FmcPubKeyY = 7,
    FmcTci = 8,
    OwnerPubKeyHash = 9,
}

pub enum ColdResetEntry4 {
    FmcSvn = 0,
    FmcLoadAddr = 1,
    FmcEntryPoint = 2,
    VendorPubKeyIndex = 3,
}

rusty1968, 2023-04-03:
@FerralCoder , @vsonims

rusty1968, 2023-04-07:
@bluegate010 , let's chat about this on slacker.

jhand2, 2023-04-10:
I see a couple possible solutions:

1. Make every dv index in the FHT larger to hold the type and data information
2. Merge the dv variants into a single enum. Make the code that handles them understand what reset domain they're in.

I think 2 is reasonable.

also cc @mhatrevi for input, who wrote much of the data vault driver

rusty1968, 2023-04-10:
Current implementation sizes indexes (storage address identifiers) as 8-bit. 

As per discussion during the meeting,  we will increase the size of indexes (re-name them to handles) and have an abstraction layer StorageLocator that understands the encoding format of the data handle and can return the information to the requester.

pub struct DataHandle(u32);
pub const FHT_INVALID_DATA_HANDLE: DataHandle = DataHandle(u32::MAX);

rusty1968, 2023-04-21:
@jhand2 , @mhatrevi , @FerralCoder  : ready for review.

FerralCoder, 2023-05-01:
closes issue: https://github.com/chipsalliance/caliptra-sw/issues/159