chipsalliance/caliptra-sw #137: Caliptra model for SoC Cmodel

varuns-nvidia, 2023-04-04:
NVIDIA uses an SoC C-model to do code development and validation for fullchip collateral. We would want to integrate a Caliptra model of the external registers https://ereg.caliptra.org for the boot process and the SoC ROM code to develop against.

@nstewart-amd had proposed a similar problem statement to @korran and @vsonims during December meet-up. Unclear whether there was a follow-up, so filing the ticket.

korran, 2023-04-04:
Yeah, this is on my plate. Will send out some header files for review later this week.

korran, 2023-04-17:
I have an example API in the branch:

https://github.com/korran/caliptra-sw/tree/hwmodel-c

* [caliptra_model.h](https://github.com/korran/caliptra-sw/blob/hwmodel-c/hw-model/examples/caliptra_model.h)
* [test_case.c](https://github.com/korran/caliptra-sw/blob/hwmodel-c/hw-model/examples/test_case.c) (example)

rusty1968, 2023-04-17:
Is an interface to caliptra-sw-emu model the only one requested?