chipsalliance/caliptra-sw #145: Does Runtime require a panic handler?

rusty1968, 2023-04-06:
Runtime specification seems to call for a panic handler as part of its mailbox command error handling flow, but we have tests in place to detect if panic symbol is included . 

A mailbox command can fail to complete in a couple ways

Hang/timeout which results in the watchdog firing
Unrecoverable panic
In both these cases, the panic handler will write diagnostic panic information to registers readable by the SoC, firmware will undergo impactless reset, and mailbox_data_avail will be asserted.

rusty1968, 2023-04-06:
@jhand2 , do we need to remove that panic missing test I included for runtime?

jhand2, 2023-04-06:
Right now DPE can panic, although I think we could potentially avoid that (I opened https://github.com/chipsalliance/caliptra-dpe/issues/54 to track this).