chipsalliance/caliptra-sw #150: [feature] Test impactless reset flow from runtime firmware

jhand2, 2023-04-07:
* Implement impactless update mailbox command
* Add integration test for impactless reset

jhand2, 2023-05-25:
More details:

* Relevant spec section on [Hitless Update](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#hitless-update)
* RT firmware could verifify the image before invoking reset to ensure update is actually valid

flanfly, 2023-05-31:
Hi. I'm trying to better understand what needs to be done here. Is the issue only for implementing the mailbox command itself, i.e. parse the incoming data and call an existing function? Also what's the exact algorithm for measuring the firmware update? Do I simply extend PCR 4-8 (what's PA btw?) with the whole image? In both cases can I (re)use code that implements the passive mode for signature verification?

jhand2, 2023-05-31:
> Is the issue only for implementing the mailbox command itself, i.e. parse the incoming data and call an existing function?

I think yes

> Also what's the exact algorithm for measuring the firmware update

After more thought, I think it makes more sense to just let ROM do the verification. See https://github.com/chipsalliance/caliptra-sw/pull/330

Today, ROM is just doing ECDSA verification ([here](https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/src/flow/update_reset.rs#L63)). But LMS verification is also being added.

closed 2023-08-07 by jhand2