chipsalliance/caliptra-sw #162: Report all FMC-relevant evidence in the FMC alias certificate

bluegate010, 2023-04-08:
Currently, the following information is used to derive the FMC CDI:

* Lifecycle state
* Debug lock state
* Anti-rollback disable state
* Vendor public key hash
* Owner public key hash
* Vendor ECC public key index
* FMC hash
* FMC SVN

However, only the following information is placed as evidence in the FMC alias certificate:

* FMC hash
* Owner public key hash

These two lists should be consistent. Any data used to derive the CDI, but not placed in the certificate over the resulting public key, cannot functionally be attested.

There are some options:

1. Add a third digest to the alias certificate, which represents a hash of all the data in the first bullet list. The two existing digests are present only as a matter of convenience.
2. Remove the two existing digests in favor of a single digest over all data in the first bullet list.
3. Leave the FMC hash digest in place in the cert, and repurpose the other digest to be over all data in the first bullet list.

Option 1 presents a complication: we will need some way to ferry the new hash from ROM to FMC. For the other data in the certificate, this is currently done by way of Data Vault slots. However, we are currently at capacity for Data Vault slots. We would need to modify RTL to allocate at least one more digest-sized slot to hold this new information that FMC will need to reconstitute the certificate.

Alternatively, we could stash this data in a PCR. The issue there is that the data that FMC pulls out and places in the certificate will not be SHA384(...); rather, it will be SHA384(0x0000 || SHA384(...)) - the result of a hash extend.

One other option would be to expand the handoff table structure to include this new data. There is no special reason for the data to be locked away in a Data Vault slot. If it mutates after ROM uses it, the certificate that FMC reconstitutes will not have a valid signature.

bluegate010, 2023-04-08:
Related question: In DICE, the SVN field is an ASN.1 Integer, meaning it's variable-length. If we want to represent the FMC's SVN in the cert, I see a few options:

* Do not use DICE's SVN field. Add the SVN into a hash of other values, and put the result in the cert as a digest.
* Add a stipulation that Caliptra SVNs never grow past 127.
* Rework the X.509 serialization library to allow for variable-length fields (not actually advocating for this but just noting it for completeness' sake).

varuns-nvidia, 2023-04-09:
> Related question: In DICE, the SVN field is an ASN.1 Integer, meaning it's variable-length. If we want to represent the FMC's SVN in the cert, I see a few options:

We discussed this in some off-hand call, the suggested outcome was to encode the SVN in the cert as the index of the most-significant bit of the actual SVN.

bluegate010, 2023-04-11:
Ack, that would work, so we only use a single byte to encode `log(SVN)`. Couple follow-up questions:

* If we do that, should we also represent the full SVN as part of a separate digest.
* If we do include the full SVN in a separate digest, is there sufficient utility that it makes sense to do both, or should we just stick with the separate digest.

bluegate010, 2023-04-25:
Update: we don't need to do any `log` encoding, since SVN can be represented in a single byte. PoR:

* Lifecycle_state:Unprovisioned --> Operational_flag:Not_Configured
* Lifecycle_state:Manufacturing --> Operational_flag:Not_Secure
* Debug lock state --> Operational_flag:Debug
* Anti-rollback disable state --> Operational_flag:Replay_Protected
* Vendor public key hash --> TCI digest
* Owner public key hash --> TCI digest
* Vendor ECC public key index --> (we can omit this, and just hash the actual vendor public key used, rather than the hash-of-hashes)
* FMC hash --> TCI digest
* FMC SVN --> We encode both SVN and MIN_SVN using a byte each. Include a high bit that's always set, so the 3-byte integer is fixed-width.

bluegate010, 2023-04-27:
Regarding anti-rollback disable: rather than representing this as an actual bit, we can instead simply suppress evidence of SVN if rollback is disabled. Presence of SVN in DICE evidence can be an implicit signal that rollback is enforced.

varuns-nvidia, 2023-05-02:
> FMC SVN --> We encode both SVN and MIN_SVN using a byte each. Include a high bit that's always set, so the 3-byte integer is fixed-width.

Jeff, is the intent to make SVN a 24-bit value? Was under impression SVN is an 8-bit value, and we must always set the MSB as 1 to make it fixed length, limiting us to 128 values (0 to 127)

bluegate010, 2023-05-02:
Ah, we could do that, in my comment above I was imagining that we'd concatenate SVN and MIN_SVN, then tack on a final bit for the MSB that gives fixed-width encoding.

But we could always just set MSB=1 on both SVN and MIN_SVN before concatenating the two of them, leading to a 16-bit value.

varuns-nvidia, 2023-05-02:
No preference from me. Do you think we'd request MIN_SVN to be a separate field in a future TCG spec?

closed 2023-09-12 by mhatrevi