chipsalliance/caliptra-sw #167: Warm Reset can lead to undefined state

varuns-nvidia, 2023-04-12:
(Starting in RTL repo, but outcome could lead to transfer to SW repo)

Warm reset cannot be pre-conditioned by Caliptra SW. Warm Reset does not clear RAMs, reset Key Vault, or reset Data Vault. Caliptra spec does not bound when a Warm Reset can occur. Therefore a Warm Reset could leave Caliptra in an undefined state.

@bharatpillilli @vsonims @jhand2 @bluegate010 @korran

varuns-nvidia, 2023-04-14:
Transferred to SW repo and assigned to @jhand2 who has action items
* determining whether runtime can continue to behave on surprise warm reset (use a debug key instead)
* @bluegate010 what journeys to be kept

jhand2, 2023-04-14:
Ack. Thanks Varun.

varuns-nvidia, 2023-05-31:
@FerralCoder the FMC spec needs updating to handle Warm Reset as well. My expectation is:
* FMC checks whether the reset type is warm_reset.
** If yes, FMC checks whether CDIFMC and PrivateKeyFMC are locked.
*** If yes, proceed to RT boot.
*** If no, fatal error.

bharatpillilli, 2023-06-01:
> @FerralCoder the FMC spec needs updating to handle Warm Reset as well. My expectation is:
> 
> * FMC checks whether the reset type is warm_reset.
>   ** If yes, FMC checks whether CDIFMC and PrivateKeyFMC are locked.
>   *** If yes, proceed to RT boot.
>   *** If no, fatal error.

Hi @varuns-nvidia 

I thought we said we will clear the key vault and continue the boot so that we don't stop the platform from booting @bluegate010  also to keep me honest.

varuns-nvidia, 2023-06-01:
@bharatpillilli , yes for RT. Should we do the same for FMC? I suppose we can still support measurement storage.

jhand2, 2023-06-02:
Added details about this in https://github.com/chipsalliance/caliptra-sw/pull/330

If we detect possible corruption, we'll call DISABLE_ATTESTATION. All commands will still work, but the key will not be endorsed by any trusted CA (since the CDI's are cleared from KeyVault).

FerralCoder, 2023-09-13:
is more work still needed here or can this be closed?

sunguk-bin-fad, 2025-02-09:
@bharatpillilli How can the SoC determine that Caliptra is in an idle state? (Can this be checked through `mbox_status` or `CPTRA_FLOW_STATUS`?) I wonder what's the safe method to trigger a warm reset.

bharatpillilli, 2025-02-10:
@sunguk-bin-fad : CPTRA_FLOW_STATUS[mbox_flow_done]. 

Please note that Caliptra at run-time doesn't generate any transactions towards SOC by its own. Only if SOC has asked for it to do some service, it is working on that service.

sunguk-bin-fad, 2025-02-27:
@bharatpillilli I have a follow-up question.
I'm using RTL1.1 on our own FPGA. (using the code - libcaliptra/example/generic) 
Based on my experiment, I've never seen that CPTRA_FLOW_STATUS[mbox_flow_done] set to 1. I'm not quite sure if monitoring this bit is the right way to determine whether Caliptra is idle. 

Simply, I checked it right after the CPTRA_FW_LOAD completion, and at that point, I believed all mailbox operations should have been completed and Caliptra should be idle, but mbox_flow_done wasn't set to 1. (https://github.com/chipsalliance/caliptra-sw/blob/caliptra-1.1/libcaliptra/examples/generic/main.c#L157).

I have monitored the bit at several other points, but it never changes. What event triggers this bit to transition to 1? Also I'm curious if (mbox_flow_done == 1) is right condition to determine whether Caliptra is idle.

activebitbit, 2026-08-17:
@sunguk-bin-fad  @bharatpillilli 
#4114  please look? thanks