chipsalliance/caliptra-sw #183: What is IDEVID_CDI_KEY?

hiption, 2023-04-15:
The Caliptra Spec.0.7 describes `UDS-HASH` to generate `IDEVID`, but ROM README,md  uses `UDS` and `IDEVID_CDI_KEY` as HMAC inputs. Where should `IDEVID_CDI_KEY` be stored? What is the difference with `UDS`?
Thank you.

hiption, 2023-05-10:
https://github.com/chipsalliance/caliptra-sw/blob/d397651349ef928d31fd6a47651457ea3f8c5930/rom/dev/src/flow/cold_reset/idev_id.rs#L35-L39

Is the `IDEVID_CDI_KEY` random value or a reference value?

bluegate010, 2023-05-10:
@vsonims can you confirm, but I believe this is simply a nothing-up-my-sleeves random number intended only to condition the raw UDS before use in DICE computation. Future-proofing; could just as easily be all-zeroes.

bluegate010, 2023-09-12:
These constants were removed in https://github.com/chipsalliance/caliptra-sw/pull/423

closed 2023-09-12 by bluegate010

bluegate010, 2023-09-12:
Filed https://github.com/chipsalliance/caliptra-sw/issues/761 to track update of the diagram.