chipsalliance/caliptra-sw #186: Status Command Needs to have FIPS Related Information

JohnTraverAmd, 2023-04-20:
FIPS requires the following information:

- Version
- Mode of operation
- Status Indicator of APIs

This may require a new HW register or an expansion of the status command.

Details for this will be completed when we get FIPS consultation.

andreslagarcavilla, 2023-04-20:
Can you file the caliptra-hw issue as well. Thanks

JohnTraverAmd, 2023-04-20:
I can, however, this could possibly be done entirely in the Mailbox command.  I was stating that I'm not sure yet on this.

I'll file the HW issue and if we get to reject it then Yay!

JohnTraverAmd, 2023-06-22:
For Notes related to the requirement:

- The cryptographic module shall output the name or module identifier and the versioning information that can be correlated with a validation record (e.g. hardware, software, and/or firmware versioning information).
- Versioning information (AS04.13) shall be provided for 
  - the physical single-chip including any excluded functional subsystem firmware (shall be specified in the OE field of the validation)
  - the sub-chip cryptographic subsystem soft and hard circuitry cores, and 
  - the associated firmware.

JohnTraverAmd, 2023-06-22:
Because we are looking into having the ROM perform crypto functionality for the user, the ROM will have to perform this command.

JohnTraverAmd, 2023-07-12:
Return Values

- Module Name ASCII = (Calitpra RTM)
- FIPS Version (HW Version | ROM Version | FW Version)
  - HW Version comes from https://chipsalliance.github.io/caliptra-rtl/main/external-regs/?p=caliptra_top_reg.generic_and_fuse_reg.CPTRA_HW_REV_ID
  - ROM Version comes from https://chipsalliance.github.io/caliptra-rtl/main/external-regs/?p=caliptra_top_reg.generic_and_fuse_reg.CPTRA_FW_REV_ID%5B0%5D
  - FW Version comes from https://chipsalliance.github.io/caliptra-rtl/main/external-regs/?p=caliptra_top_reg.generic_and_fuse_reg.CPTRA_FW_REV_ID%5B0%5D
- Mode of operation
  - Boolean FIPS Approved/Non-Approved (Defaulted to FIPS Approved)

rusty1968, 2023-07-20:
The ROM version information should be written to the first word of the array by ROM  and runtime should populate the second word. Any objections?

JohnTraverAmd, 2023-07-20:
I think that works.

closed 2023-09-13 by JohnTraverAmd