chipsalliance/caliptra-sw #211: Architecting ROM for Validation

rusty1968, 2023-04-27:
Problem statement :  A fully featured ROM  takes a long time to simulate. We need the ability to selectively skip certain flows depending on the test scenario. 

An approach would be to perform the selectively skip certain flows via fuse bits. From the security perspective, this is not ideal. Conditional compilation is an improvement, but is still intrusive and makes the code harder to maintain.

A superior approach would be to enable Caliptra  integration validation of ROM by using mocking technique. This is a very popular approach in TDD methodology. 

To apply mocking

rusty1968, 2023-04-27:
@fdamato , @attzonko

rusty1968, 2023-04-27:
@korran

attzonko, 2023-04-27:
So are you thinking of achieving this by adding traits to ROM modules which don't have an impl currently?

korran, 2023-04-27:
Is simulation necessary? Why not just use an FPGA?

rusty1968, 2023-04-27:
Hi, @korran
RTL simulation is bread and butter for design verification teams. The FPGA validation vehicle will be useful for software development.