chipsalliance/caliptra-sw #228: Can we free up DataVault slots that hold FMC/RT TCI?

bluegate010, 2023-05-02:
With the submission of https://github.com/chipsalliance/caliptra-sw/pull/196, FMC no longer needs to consult DataVault to reconstitute the TBS for FMC's alias certificate. Currently, there are two slots allocated for TCIs that are used in DICE computation and in assembling the cert: FMC TCI and RT TCI.

RT TCI _may_ still be needed because it is not used to permute the CDI until FMC runs. If it is not in DataVault it would need to be passed to FMC via DCCM. FMC TCI, on the other hand, is purely informational: it was already used by ROM to tweak the CDI before control passes to FMC. If the FMC TCI is corrupted, the worst case is a DOS.

ajisaxena, 2023-05-19:
As we discussed in the meeting on 05/18, the FMC is still using this and we will have to move the FMC (not the test FMC) to start using the TCI from DCCM rather than data vault. The original intention of  #196 was to make integration test easy to implement but we realized later that the same approach can be taken for actual implementation. We will keep an eye on this and once the FMC is moved to using the TCI from DCCM we will address this issue.

jhand2, 2023-05-19:
I'm inclined to say we can remove FMC TCI but should probably leave RT TCI since it isn't just informational. I'll raise this in slack.

FerralCoder, 2023-05-19:
I’m thinking we can just modify the FHT handle def so that it can refer to DCCM in addition to the vaults. Then we could keep both items in FHT with full flexibility to move them to any storage location we want

bluegate010, 2023-07-15:
Upon review I actually don't think we can free up the FMC TCI slot, because that is used to ensure that the FMC does not change across hitless update.

However, I _do_ think we can free up the slots that are holding the LDevID and FMC alias public keys. The LDevID slot ID is not even present in the FHT, and the FMC alias pub key is only used by FMC to validate the signature it generates over RT's alias cert. If FMC needs the public key it can always regenerate it from the private key. Plus we'll be taking care of those consistency checks within the driver anyways.

FerralCoder, 2023-09-13:
can this be closed?

closed 2023-09-18 by mhatrevi