chipsalliance/caliptra-sw #238: Update ROM spec to correct owner key flow

bluegate010, 2023-05-04:
As identified by @varuns-nvidia in https://github.com/chipsalliance/caliptra-sw/issues/222#issuecomment-1530770019.

The ROM spec has a [flowchart](https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/README.md#preamble-validation-steps) that includes details on owner key verification. These details should be updated:

* If owner key fuses are not populated, the owner key is not compared with them.
  * Many devices will not have these fuses populated, to support circular economy.

In addition, we should consider allowing an edge where the image has no owner key, in which case no such verification is performed.

jhand2, 2024-02-21:
@mhatrevi should we change this for the 1.0 spec?

bluegate010, 2024-02-21:
> In addition, we should consider allowing an edge where the image has no owner key, in which case no such verification is performed.

Just noting that this edge was never implemented; the image must always be signed by an owner key.

closed 2024-02-23 by jhand2