chipsalliance/caliptra-sw #281: SW version (ROM and FMC) for RTL tag v0.8

morris-c-mao, 2023-05-17:
Hi, we integrated RTL tag v0.8 into our SoC.
Could we just follow the latest SW (ROM and FMC) or which commit is better for RTL tag v0.8?

morris-c-mao, 2023-05-31:
Can we use ROM v0.5.2 for RTL v0.8?
And does commit "64b30c0" present ROM v0.5.2?

FerralCoder, 2023-09-13:
@fdamato, please respond to this

attzonko, 2023-09-13:
At the time the RTL was tagged as v0.8 there was no methodology for providing a coherent SW (ROM+FMC/RT) for that release. Since then the team created a coherent nightly release process. However it doesn't go backwards to RTL v0.8, so I am not sure it would be an achievable task to find a version of SW which works with RTL tagged v0.8. My recommendation is to use the known working coherent releases: https://github.com/chipsalliance/caliptra-sw/releases

fdamato, 2023-09-13:
Just to adds up, the release process Alex is mentioned is documented in https://github.com/chipsalliance/Caliptra/blob/main/HWReleaseProcess.md

closed 2023-09-13 by FerralCoder