chipsalliance/caliptra-sw #307: ROM should write out the TBS length to DCCM

bluegate010, 2023-05-23:
ROM currently [stages](https://github.com/chipsalliance/caliptra-sw/blob/2e26d715dcbfa92c1e5e7eb3bcd2cda760db9792/rom/dev/src/flow/cold_reset/mod.rs#L76) DICE cert TBS portions in DCCM. However, it does not include the length. This could pose FMC/RT compatibility problems across ROM generations.

One thing to note: any code that reads this length field should defensively ensure it is not longer than the DCCM region bounds for that cert, i.e. LDevID's TBS length is not larger than `FMCALIAS_TBS_ORG - LDEVID_TBS_ORG`.

jhand2, 2023-05-23:
On possibility would be to add the length to the hand-off table

jhand2, 2023-05-24:
Another suggestion, maybe we should just have ROM populate the HOT with both the address and size for each cert. That way different ROM versions can put these in different places and FMC/RT can tolerate those changes.

mhatrevi, 2023-06-02:
Duplicate of https://github.com/chipsalliance/caliptra-sw/issues/343

closed 2023-06-02 by mhatrevi