chipsalliance/caliptra-sw #314: Run wycheproof tests against Caliptra

jhand2, 2023-05-25:
* [ECDSA verify tests](https://github.com/google/wycheproof/blob/master/testvectors_v1/ecdsa_secp384r1_sha384_test.json) can be run using Caliptra's ECDSAVerify command
* [HMAC](https://github.com/google/wycheproof/blob/master/testvectors_v1/hmac_sha384_test.json) do not currently have a mechanism for external testing of the block. But might be needed for FIPS anyway? Otherwise, could build a test mailbox command for testing.

flanfly, 2023-05-31:
Just to clarify: the issue is only about running the Known Answer Tests in the wychenproof repository against the hardware implementations? Do you guys imagine this as fully integrated system tests (i.e. `cargo test -- wychenproof`)?

ArthurHeymans, 2023-08-08:
Completed

jhand2, 2023-08-08:
Yep, thanks!

closed 2023-08-08 by jhand2