chipsalliance/caliptra-sw #317: Boot time profiling

jhand2, 2023-05-25:
We should do profiling of how long Caliptra boot takes and collect metrics on which part(s) of boot take the longest. This will help identify areas for optimization to lessen Caliptra's impact on overall SoC boot time.

flanfly, 2023-06-01:
Do you imagine this profiling to be done on the actual hardware or on the simulator? I'm asking because I'm trying to understand how much data is going to be collected. I presume the hardware does not have enough memory to store whole traces but only a few timestamps.

jhand2, 2023-06-01:
We don't have real hardware, so it would need to be with verilator (or FPGA) probably. I think we can measure cycle count to figure out how fast it will be on real hardware.

jlmahowa-amd, 2023-07-19:
Attached log from FPGA. A couple things to keep in mind:

- Current FPGA logging mechanism introduces delays as the FW waits for the test to acknowledge each byte.
- Integration spec suggests a clock of 400MHz. Current FPGA is at 20MHz.

[fpga_log_July19.txt](https://github.com/chipsalliance/caliptra-sw/files/12100561/fpga_log_July19.txt)

andreslagarcavilla, 2023-09-08:
Adding @benjamindoron @ArthurHeymans @nquarton @ericeilertson @attzonko 

As discussed, can we 
* fork the boot smoke test into one using an invalid LMS key that will consume the most time
* characterize boot timing cycles for "normal" and worst case scenarios

Benjamin and Arthur, can you help Nick and Alex from AMD and Eric from Microsoft, land this?

I'm not sure who to assign ownership to within this group. Thanks

nquarton, 2023-09-12:
Here are the full logs for the boot with LMS enabled for both the valid key and an invalid key. The timing isn’t all that different, but the invalid key is a bit slower as expected.

This is just the smoke test with minor modifications:
1.	Turned on LMS verify (this will be committed in a PR soon)
2.	Logging around the LMS verify calls and image verify in general to get the timing data
3.	The invalid LMS case was modified to ignore the LMS verify failure and continue to boot.

Invalid LMS key full log
 
```
      6,155 ready_for_fuses is high
      6,329 writing to cptra_bootfsm_go
test smoke_test has been running for over 60 seconds
    195,166 UART: 
    195,209 UART: Running Caliptra ROM ...
    196,318 UART: 
    196,881 UART: [state] CFI Enabled
    198,267 UART: [state] LifecycleState = Production
    199,931 UART: [state] DebugLocked = Yes
    202,189 UART: [state] Starting the Watchdog Timer
    346,030 UART: ROM Digest: ABD791D9580684A694D2C76F8FD092C362B3A10888D1186EF51B89750A7346D3
    356,595 UART: [kat] ++
    357,023 UART: [kat] sha1
    366,191 UART: [kat] SHA2-256
    368,970 UART: [kat] SHA2-384
    373,031 UART: [kat] SHA2-384-ACC
    375,146 UART: [kat] ECC-384
  3,943,214 UART: [kat] HMAC-384
  3,953,021 UART: [kat] LMS
  5,009,096 UART: [kat] --
  5,012,051 UART: [cold-reset] ++
  5,014,013 UART: [idev] ++
  5,014,465 UART: [idev] CDI.KEYID = 6
  5,015,445 UART: [idev] SUBJECT.KEYID = 7
  5,016,585 UART: [idev] UDS.KEYID = 0
  5,034,986 UART: [idev] Erasing UDS.KEYID = 0
  8,297,173 UART: [idev] Using Sha1 for KeyId Algorithm
  8,319,582 UART: [idev] --
  8,323,451 UART: [ldev] ++
  8,323,917 UART: [ldev] CDI.KEYID = 6
  8,324,887 UART: [ldev] SUBJECT.KEYID = 5
  8,326,041 UART: [ldev] AUTHORITY.KEYID = 7
  8,327,293 UART: [ldev] FE.KEYID = 1
  8,338,135 UART: [ldev] Erasing FE.KEYID = 1
 11,617,678 UART: [ldev] Signing Cert with AUTHORITY.KEYID = 7
 15,210,901 UART: [ldev] PUB.X = 842C00AF05ACCCEB14514E2D37B0C3AAA218F15057F1DCB824A214980B744688A0888A0297FA7DC5E1EAD8CA1291DB22
 15,225,744 UART: [ldev] PUB.Y = 9C28EB8678BCE800822C07228F416AE49D218E5DA2F2D1A8A27DC19ADF668A74628999D222B40159D8076FAFBB8C5EDB
 15,240,942 UART: [ldev] SIG.R = 0C1B95861ADA0DF572438E88B23DDD5B2C24C974DF359988CC54E39E3145635A94E3D8196B49164CAD991714C2B18892
 15,255,768 UART: [ldev] SIG.S = 4BEDF2A7DEC9405964ADF4949819FADE2CD785CD0307078F74EA4E5CDDF8D1FAE5380507AB09871344CA288FA6961532
 15,276,081 UART: [ldev] --
 15,283,901 UART: [state] Stopping the Watchdog Timer
 15,287,093 ready_for_fw is high
 15,287,095 <<< Executing mbox cmd 0x46574c44 (53148 bytes) from SoC
 15,287,135 UART: [afmc] Waiting for Commands...
 15,313,888 UART: [afmc] Received Image of size 53148 bytes
 15,316,455 UART: [state] Starting the Watchdog Timer
 15,377,669 UART:                 ---Verify start---
 16,678,302 UART: LMS verify start
 16,679,033 UART: 
 18,822,987 UART: LMS verify complete
 18,823,847 UART: 
 20,115,541 UART: LMS verify start
 20,116,272 UART: 
 22,122,872 UART: LMS verify complete
 22,123,732 UART: 
 22,165,436 UART:                 ---Verify complete---
 22,166,490 UART: [afmc] Image verified using Vendor ECC Key Index 0
 22,320,063 UART: [afmc] Loading FMC at address 0x40000000 len 12672
 22,398,936 UART: [afmc] Loading Runtime at address 0x40004000 len 34592
 22,611,323 >>> mbox cmd response: success
[... 141 lines cut ...]
 29,128,759 UART: [alias rt] Populate DV Done
 29,129,644 UART: Handoff : FMC CDI: 6
 29,130,312 UART: FMC Alias Private Key: 7
 29,132,126 UART: [alias rt] Derive CDI
 29,132,792 UART: [alias rt] Store in in slot 0x4
 29,166,294 UART: [alias rt] Derive Key Pair
 29,167,115 UART: [alias rt] Store priv key in slot 0x5
 32,405,664 UART: [alias rt] Derive Key Pair - Done
 32,416,567 UART: [alias rt] Signing Cert with AUTHO
 32,417,617 UART:             RITY.KEYID = 7
 34,695,450 UART: [alias rt] Erasing AUTHORITY.KEYID = 7
 34,696,831 UART: [alias rt] PUB.X = B0F69BF99973255C109F0890F9BC901830F62509DDCBE845F15BA0120E0D1917BFB79CF033126E93EAB8367297A71711
 34,703,821 UART: [alias rt] PUB.Y = 331734EF4DADB80206A4C68778F401D93910D009F92817987E2A5124DFD8AC51A71BE7914DD3411C156B051402BD64F7
 34,710,985 UART: [alias rt] SIG.R = 2EA73FF07617F7E3EE6FF2616680D525359DBFC479E69B8C924FA5870E46E1B1858101B4B933249C3B27E27702364AF4
 34,717,952 UART: [alias rt] SIG.S = E0BFA74D6044D9FB824D6AD599CBC5481BF26E4C25F69E1631D634D1C67457138B778E884F5840C40DDABD4B8CB4B220
 36,019,146 UART: 
 36,019,176 UART:   ____      _ _       _               ____ _____
 36,020,646 UART:  / ___|__ _| (_)_ __ | |_ _ __ __ _  |  _ \_   _|
 36,022,146 UART: | |   / _` | | | '_ \| __| '__/ _` | | |_) || |
 36,023,586 UART: | |__| (_| | | | |_) | |_| | | (_| | |  _ < | |
 36,025,026 UART:  \____\__,_|_|_| .__/ \__|_|  \__,_| |_| \_\|_|
 36,026,466 UART:                |_|
 36,027,042 UART: 
 36,123,074 UART: Caliptra RT listening for mailbox commands...
```

attzonko, 2023-09-13:
@andreslagarcavilla @benjamindoron @ArthurHeymans @korran @ericeilertson 
Is the data from Nick sufficient for the information needed? What else is needed from us to close this issue?

andreslagarcavilla, 2023-09-13:
thanks everyone, this is cool. Do I understand correctly that it takes 36 million cycles to get to the mailbox loop? ~90 millis @400Mhz?

nquarton, 2023-09-13:
That is my understanding. I've heard the 80-90ms time quoted a few times. (But do note there is also limited command handling in ROM much earlier in the boot for certain scenarios.)

ericeilertson, 2023-09-25:
To ensure the maximum verification time requires the candidate OTS signature generation to always start at 0 in the winternitz chain, thus forcing each of our 48 hash chains to always be 16 long.  To do this we need to change the implementation of the coefficient function in the LMS driver to always return 0.  I'll chat with @korran on how to monkey patch the coefficient function for this test.

closed 2023-12-01 by mhatrevi