chipsalliance/caliptra-sw #321: FIPS MB Command to Run and Report on Self Tests

JohnTraverAmd, 2023-05-25:
Requirement:
Enable a MB FIPS command which will perform all required FIPS Self-Tests.  This includes:

1. ROM Self Test
2. FW Authentication
3. FW Hash Checks from authenticated manifest
4. All KAT tests
5. RNG Self-Tests

If any self test fails, the MB response will provide the failure AND the overall Caliptra will not respond to any other operation request.

JohnTraverAmd, 2023-06-22:
Because we are looking into having the ROM perform crypto functionality for the user, the ROM will have to perform this command.

rusty1968, 2023-07-27:
@korran  raised the issue of LMS and SHA-1 being implemented in firmware and we having no guarantee that RT and ROM will be using the same version.

closed 2023-09-08 by korran