chipsalliance/caliptra-sw #324: All Temp Data Must be Zeroized on Completion of Operation

JohnTraverAmd, 2023-05-26:
Requirement: All Temp Data Must be Zeroized on Completion of Operation

Details:  Looking into rust, if we can prove that all data is zeroized upon garbage collection, then this will meet the requirement.  We will have to prove that this isn't disabled or able to work around.
Otherwise, before we leave scope for any intermediate data related to crypto operations, must be zeroized.

jhand2, 2023-05-26:
Is this for all crypto-related data (e.g. public keys, signatures) or just all secret data? If the latter, it would just be a matter of clearing any intermediate keyvault slots I think.

JohnTraverAmd, 2023-06-01:
This is related to crypto operations.  If RUST doesn't automatically do this, we will need to zeroize it in the code before we exit the functions.

JohnTraverAmd, 2023-06-01:
Each crypto function should be reviewed by Vishal M.

JohnTraverAmd, 2023-06-22:
https://github.com/chipsalliance/caliptra-sw/pull/373

korran, 2023-06-23:
With modern C or Rust compilers, it's pretty much impossible to guarantee that the compiler hasn't made a temporarily copy of these values somewhere where it won't get cleared (cpu registers, etc).

Can you point us to the actual requirements? Guaranteeing all temporaries are cleared is pretty much impossible without writing our own compiler.

JohnTraverAmd, 2023-06-23:
What Vishal has done is going to be sufficient.  We need to make an effort to zeroize, and he has put in code to zeroize HW data and the end data on crypto algos.

The requirement applies to crypto metadata related to the algorithm.

The requirement is 9.32: Temporary SSPs shall be zeroised when they are no longer needed.

swkeypair, 2023-06-23:
> Can you point us to the actual requirements as documented by NIST?

_ISO 24759 (Derived Test Requirements) AS09.32 re **Sensitive security parameter zeroization**_

> AS09.32: Temporary SSPs shall be zeroised when they are no longer needed.
> VE09.32.01: The vendor documentation shall specify that temporary SSPs are zeroised when they are no longer needed.
> TE09.32.01: The tester shall verify that the vendor provides documentation specifies that temporary SSPs are zeroised when they are no longer needed.

This is a Level 2+ requirement newly introduced in the FIPS 140-3 era, and as such there are grey areas, like zeroization of hardware registers. Consider this as a good hygiene / best practices requirement; so far this has not been an absolute.
A public example implementation of an aspect of this can be found in the [javacard.security library key interface](https://docs.oracle.com/javacard/3.0.5/api/javacard/security/Key.html) and related subinterfaces.

swkeypair, 2023-06-23:
That last reference was a bit terse:

> A public example implementation of an aspect of this can be found in the [javacard.security library key interface](https://docs.oracle.com/javacard/3.0.5/api/javacard/security/Key.html) and related subinterfaces.

To explain: in OO languages, use of a destructor that performs zeroization can be convenient for to at least partially address this item.

jhand2, 2023-06-23:
From the text above it's still not clear to me that public cryptographic metadata (such as public keys and signatures) are intended to be considered "sensitive security parameters". Can you confirm this interpretation?

swkeypair, 2023-06-23:
The standards docs lump public keys into this, but in another requirement, provide a way to indicate when public keys are exceptions to destruction practices. It is not at all clear to me what the concern is regarding the public keys; more likely it is simply an artifact of renaming such keys in the newest version of the standard.

For the record and as briefly as possible, the revised terminology in ISO 19790:

> **sensitive security parameters (SSP):** critical security parameters (CSP) and public security parameters (PSP)

> **critical security parameter (CSP):** security related information whose disclosure or modification can compromise the security of a cryptographic module
**EXAMPLE** Secret and private cryptographic keys, authentication data such as passwords, PINs, certificates or other trust
anchors
**NOTE** A CSP can be plaintext or encrypted.

> **public security parameter (PSP):** security related public information whose modification can compromise the security of a cryptographic module
**EXAMPLE** Public cryptographic keys, public key certificates, self-signed certificates, trust anchors, one time passwords
associated with a counter and internally held date and time
**NOTE** A PSP is considered protected if it cannot be modified or if its modification can be determined by the module.

The basic idea of AS09.32 is to destroy keys and secrets when they are no longer in use.
If a key pair is instantiated and goes out of scope, and there is no need to retain the public key, then destroy them both.

closed 2023-06-30 by korran