chipsalliance/caliptra-sw #325: All MB Commands will have a FIPS Status Indicator Return Field

JohnTraverAmd, 2023-05-26:
Requirement: All MB Commands will have a FIPS Status Indicator Return Field

Details: Each return message will have a 32 bit return value which will indicate if the FIPS operation was approved/non-approved/not-applicable or a FIPS error indicator.

JohnTraverAmd, 2023-06-01:
1st step will be to update the spec to include the new return values.

Vishal M will work on the ROM supported messages.  The rest Jordan can own.

JohnTraverAmd, 2023-06-23:
Types of Return Values:

- FIPS Boot Approved
- FIPS RTM Approved
- FIPS Non-Approved

If the service has no cryptographic functionality, it could still return FIPS Approved.  This would still be listed in the Security Policy.

When we get an error within the module, those failures would be return in the main error code.  This field would still indicate approved.

nquarton, 2023-09-07:
FIPS status responses in runtime added by https://github.com/chipsalliance/caliptra-sw/pull/534

Support for FIPS status in ROM responses added by https://github.com/chipsalliance/caliptra-sw/pull/627 and https://github.com/chipsalliance/caliptra-sw/pull/658.

https://github.com/chipsalliance/caliptra-sw/pull/734 adds the FIPS status response to Vishal M's pending stash measurement addition in ROM.

closed 2023-09-12 by JohnTraverAmd