chipsalliance/caliptra-sw #326: FIPS MB Command to Zeroize All Data/Keys

JohnTraverAmd, 2023-05-26:
Requirement: FIPS MB Command to Zeroize All Data/Keys

Details:
A MB command is required to support zeroizing of all keys and data within the FIPS module.

JohnTraverAmd, 2023-06-23:
https://github.com/chipsalliance/caliptra-sw/issues/324

JohnTraverAmd, 2023-06-23:
Because we are investigating ROM services in a FIPS boot mode, ROM needs to support this message.

JohnTraverAmd, 2023-06-23:
This command will need to support EOL vs a power off case.

JohnTraverAmd, 2023-07-12:
Expected behavior:

- Clear Data Vault
- Clear data memory (handoff table, etc.)

Send a completion response.
Then, enter an error loop which rejects all commands with error.

A test which confirms that we are sending errors on further commands.
This may be the disable attestation command - https://github.com/chipsalliance/caliptra-sw/tree/main/runtime#disable_attestation

JohnTraverAmd, 2023-07-12:
@jhand2 do you think this requirement is fulfilled in the https://github.com/chipsalliance/caliptra-sw/tree/main/runtime#disable_attestation command?

jhand2, 2023-07-12:
One note about clearing data memory: If we do this, this will also prevent us from doing any hitless/warm resets. Once FHT, certs, etc are cleared, we can only fully repopulate them on cold boot.

This might be fine, just want to call that out.

JohnTraverAmd, 2023-07-12:
I see.  That makes sense.  

I think that could be the difference between these two commands.  The FIPS Zeroize command would NOT want to support a Hitless FW update and be able to continue.  Its goal is to ensure all internal data is zeroized.  We can have both commands and have different functionality there, I just didn't want to duplicate the same behavior.

rusty1968, 2023-07-25:
Some data vault entries are locked by the ROM. We can't erase all entries in RT.

rusty1968, 2023-07-26:
@jhand2 , we need to acknowledge the completion of shutdown via maiilbox and write to non-fatal error (if fatal error register is better than we can do that). guess it is report_error_and_halt then.

closed 2023-09-08 by korran