chipsalliance/caliptra-sw #345: Setting ROM region as cacheable in MRAC - Cause RISC-V to throw an exception on FPGA

fdamato, 2023-06-02:
On FPGA, setting ROM region as cacheable in MRAC causes RISC-V to throw "invalid operation" exception. 
When the exception is triggered, the MEPC register points to the PC location where the the MRAC register is set.
While, the issue doesn`t present on Verilator, an investigation should be performed to rule out an actual RTL cause.

jlmahowa-amd, 2023-06-06:
We are likely running into FPGA specific issues.
When the illegal instruction exception is fired mtval gets stored with the instruction.
mtval: 0x**a0b7**9073
Supposed to be: 0x**7c00**9073

A slightly different ROM binary with identical setup code running on the same FPGA image does not reproduce this illegal instruction exception.

jlmahowa-amd, 2023-07-10:
This was resolved by fixing the FPGA frequency.

closed 2023-07-10 by jlmahowa-amd