chipsalliance/caliptra-sw #355: ROM code size exceeds 32KB

ChiaweiW, 2023-06-06:
Hi All,

I used the build.sh script to construct Caliptra ROM binary.
However, the binary size is 72KB, which excceds the ROM code size 32KB described in the HW spec. (as shown in the link)
[Caliptra_HW_Block_diagram.png](https://github.com/chipsalliance/Caliptra/blob/main/doc/images/Caliptra_HW_Block_diagram.png)

Did I miss something during the build? Any guidance is appreciated.

Regards,
Chiawei

jhand2, 2023-06-06:
The output binary is an ELF file. Only the `.text` and `.rodata` sections actually make up the ROM image size (stack, bss, etc occupy DCCM RAM).

You can see the sections and sizes with `readelf -S caliptra-rom`

ChiaweiW, 2023-06-06:
After `make objcopy` under rom/dev/, the generated binary is padded to exactly 32KB.
Thanks a lot for your help.

closed 2023-06-06 by ChiaweiW