chipsalliance/caliptra-sw #356: Consider moving all long-lived DCCM structures to one region

jhand2, 2023-06-06:
Currently, the linker script specified several regions for storing certs, PCR log, fuse log, etc. in DCCM. The suggestion is to instead have one region in DCCM for the FHT and move all of those other things into that struct.

For example:

```rust
struct HandoffTable {
  ...
  ldevid_cert: [u8; MAX_CERT_SIZE];
  fmc_alias_cert: [u8; MAX_CERT_SIZE];
  rt_alias_cert: [u8; MAX_CERT_SIZE];
  fuse_log: [FuseLogEntry; MAX_FUSE_LOG_ENTRIES];
  pcr_log: [PcrLogEntry; MAX_PCR_LOG_ENTRIES];
  ...
}
```

cc @mhatrevi @rusty1968 @korran

FerralCoder, 2023-06-06:
I think this proposal makes sense.

rusty1968, 2023-06-09:
I will take this one.

rusty1968, 2023-06-13:
@mhatrevi , @jhand2  : this requires some refactoring as now we are instantiating make_fht in the stack - maybe we need to get together and make sure we have enough tests to make this a safe refactor.

rusty1968, 2023-06-13:
@korran  , @vsonims

jhand2, 2023-06-21:
+1 to that recommendation

korran, 2023-11-17:
Everyone is happy with the solution in https://github.com/chipsalliance/caliptra-sw/blob/main/drivers/src/persistent.rs

closed 2023-11-17 by korran