chipsalliance/caliptra-sw #391: HW Release: Caliptra C API package provided as part of the supporting ZIP File

fdamato, 2023-06-15:
The Caliptra C API Library should be distributed in the collateral supporting ZIP File, as part of the HW Release.
The provided APIs should allow the SOC to interact with Caliptra, through (included but not limited to) mailbox communication.
The library should need to be compiled by the integrator in their environment, who should need to:
          - Implement read/write functions which will perform the SOC-specific Caliptra register access  
          - Provide toolchain cross compiler 

An example of the interface provided to the integrator can be:
    `- int caliptra_upload_fw(struct caliptra_buffer *fw_buffer);`

wmaroneAMD, 2023-07-10:
Reassigned to myself. I've linked my branch on my fork of caliptra-sw.

wmaroneAMD, 2023-08-11:
I think this is done.

closed 2023-09-13 by FerralCoder