chipsalliance/caliptra-sw #424: MB Command Parameter Checking

JohnTraverAmd, 2023-06-23:
MB Command Parameter Checking

When a MB command is not providing correct parameters, the MB will reject the command as INVALID_PARAMETERS.

rusty1968, 2023-06-24:
FMC does not handle mailbox commands. Are there any planned?

JohnTraverAmd, 2023-06-24:
No. FMC shouldn't service any mailbox commands. Just ROM and runtime.

rusty1968, 2023-06-24:
Why the FMC label?

rusty1968, 2023-06-24:
Thanks for removing it :-)

JohnTraverAmd, 2023-06-27:
Possible Errors:

- Reject for Unsupported Command
- Reject for Size
- DPE Parameter Commands Invalid if inaccurate.

Test Created that runs some invalid commands for each error case added.

rusty1968, 2023-06-29:
ROM : Firmware download command is checking parameters. Stash measurement is still in progress.

mhatrevi, 2023-09-18:
1. FIRMWARE_LOAD:  
- Checks for if dlen is not 0 or not greater than the max. image size.
- Verifier library validates the image.

2. VERSION
3. SELF_TEST_START 
4. SELF_TEST_GET_RESULTS 
5. SHUTDOWN 
6. CAPABILITIES 
7. STASH_MEASUREMENT
- Verifies checksum of the command with the value in the command header.
- Verifies dlen matches the data received.
-

closed 2023-09-19 by JohnTraverAmd