chipsalliance/caliptra-sw #445: ROM warm reset flow not implemented

jlmahowa-amd, 2023-06-30:
Because the warm reset flow skips FW loading and ldevid/fmcalias key derivation some fields of the FHT are not populated:
- manifest_load_addr
- ldevid_tbs_size
- fmcalias_tbs_size
Because FHT fields are not populated the FMC sees it as invalid and fails.

jhand2, 2023-06-30:
I think we'll want to just load the old FHT from DCCM. I've made a similar change for update reset: https://github.com/chipsalliance/caliptra-sw/pull/444

jlmahowa-amd, 2023-07-19:
I created a PR with my attempt at implementing the warm reset flow: https://github.com/chipsalliance/caliptra-sw/pull/514
@mhatrevi If I use your modifications to the warm reset flow with my test it fails with "Handoff Error: Invalid RT Entry Point 0x000F0004" because the data vault is not repopulated.

closed 2023-08-16 by mhatrevi