chipsalliance/caliptra-sw #494: ROM should track the minimum RT SVN since cold boot in a warm-reset 4-byte DV slot

bluegate010, 2023-07-17:
On cold boot, ROM currently [writes](https://github.com/chipsalliance/caliptra-sw/blob/26fb69901e49b2fe50fe1547e75558e4051edbff/rom/dev/src/flow/cold_reset/fw_processor.rs#L320) the RT SVN to a warm-reset 4-byte DV slot. This slot is currently not queried by FMC or RT FW.

On hitless update, ROM should update this field to be `MIN(current_val, new_fw.svn)`. RT FW can then query this value and attest to it. Motivation: attestation to a minimum-SVN is more consumable by autonomous verifiers (e.g. PA-RoTs or confidential-compute customers) than a PCR log, which can really only be consumed by a central verifier.

If we want the cold-boot RT SVN to be preserved so it can be attested later, we could place it in a cold-boot 4-byte DV slot. https://github.com/chipsalliance/caliptra-sw/pull/479 frees one up.

bluegate010, 2023-08-08:
It turns out there are warm-resettable 4-byte DV slots still available, so I believe this could be done in FMC instead of ROM.

rusty1968, 2023-12-01:
PR#663

closed 2023-12-01 by rusty1968