chipsalliance/caliptra-sw #518: Update RT load/entry addresses in DV during update reset

jhand2, 2023-07-19:
During update reset, need to call

```rust
venv.data_vault.write_warm_reset_entry4(WarmResetEntry4::RtLoadAddr, manifest.runtime.load_addr);
venv.data_vault.write_warm_reset_entry4(WarmResetEntry4::RtEntryPoint, manifest.runtime.entry_point);
```

This is because these values can change with newer firmware versions.

This fails today with a Store/Load Fault, presumably because the DV entries are locked? I'm not sure why they're locked though.

rusty1968, 2023-07-25:
Why do we need load addresses?

jhand2, 2023-07-25:
Discussed at standup last week. I think it is fine to just have one or the other (and require that the load address and entry point are the same).

As for why we need to update them at all, the address is read from the image header. So an update could change where the RT image gets loaded.

closed 2023-08-07 by mhatrevi