chipsalliance/caliptra-sw #523: Zero owner_pk_hash causes unexpected results

jlmahowa-amd, 2023-07-20:
This was observed on FPGA and sw emulator.
main branch at commit: ca319e19e550e923289beb7cfb100ab69e09de39

Using the smoke_test if owner_pk_hash is replaced with an invalid hash the expected result is IMAGE_VERIFIER_ERR_OWNER_PUB_KEY_DIGEST_MISMATCH when verifying the FW image. For non-zero incorrect owner_pk_hash values this is the result, but if the owner_pk_hash is zero the result is different.

On FPGA:
owner_pk_hash: [0xF; 12] results in IMAGE_VERIFIER_ERR_OWNER_PUB_KEY_DIGEST_MISMATCH.
owner_pk_hash: [0x0; 12] results in the ECC KAT to fail with the error ROM_KAT_ECC384_SIGNATURE_GENERATE_FAILURE
With a zero value of owner_pk_hash FPGA fails much earlier, during ROM execution of the KATs.

Using the sw-emulator;
owner_pk_hash: [0xF; 12] results in IMAGE_VERIFIER_ERR_OWNER_PUB_KEY_DIGEST_MISMATCH.
owner_pk_hash: [0x0; 12] results in the smoke test passing.
With a zero value of owner_pk_hash the sw-emulator makes it to RT firmware.

Using Verilator:
owner_pk_hash: [0x0; 12] Still running but can confirm that it doesn't fail at the KAT

jlmahowa-amd, 2023-07-20:
Verilator also passes the smoke test with owner_pk_hash full of zeroes.

jlmahowa-amd, 2023-07-20:
@jhand2 to confirm that a zero owner_pk_hash is acceptable. Image Verifier does explicitly only error on mismatch if the hash is non-zero.
FPGA is still a concern, we don't expect the fuses to impact the KAT.

jhand2, 2023-07-20:
I confirmed that succeeding with a zero'd owner_pk_hash is expected.

All the inputs/outputs to the ECC KAT are constant (including the private key, signature, etc). So the fuse values shouldn't impact the execution of that KAT.

I don't know a lot about FPGA synthesis, so I'm having a hard time speculating about what could be going wrong. One thing to check would be where exactly it's failing.

It could be here:

https://github.com/chipsalliance/caliptra-sw/blob/5eb58b0ae8e6430ac6cceb122eb7c028864b2442/kat/src/ecc384_kat.rs#L72

or here:

https://github.com/chipsalliance/caliptra-sw/blob/5eb58b0ae8e6430ac6cceb122eb7c028864b2442/kat/src/ecc384_kat.rs#L75

If it's the first block, then finding out where in the `sign` function it's failing and with what error code would be good.

jlmahowa-amd, 2023-07-21:
Mistake in model-fpga-realtime. Offset takes a DW offset and the byte offset was provided for one case where the TRNG_STATUS register was read. That read was actually accessing a DW from the owner_pk_hash. When owner_pk_hash was written with the valid hash or the couple of invalid values I tried bit 0 happened to be set. ECC KAT test failed to get ETRNG data.

PR #531 (2023-07-25): Correct address of trng_status register in FPGA model
Fixes #523
closed 2023-07-21 by jlmahowa-amd