chipsalliance/caliptra-sw #624: Implement structure-aware fuzzing

benjamindoron, 2023-08-10:
Rather than taking a byte slice as an argument, we can take any struct, and the input will be mutated to match the struct types. In theory, this helps pass more validation checks.

Another use case is automating the process of preparing multiple sets of function arguments. For instance, the DPE fuzzing target could fuzz the support matrix, whether this can lead to undefined behaviour.

Implementing this is a matter of making relevant structs and their internals conditionally derive `Arbitrary` from the crate that performs this mutation.

benjamindoron, 2023-08-10:
It doesn't seem like I can assign myself, though I am working on this.

benjamindoron, 2023-08-30:
Done for image verifier and LMS, I've been told to wait regarding further work.

korran, 2023-11-03:
Completed

closed 2023-11-03 by korran