chipsalliance/caliptra-sw #674: Mechanism to run self-tests and firmware authentication from ROM

korran, 2023-08-22:
FIPS requires that we have a mechanism to run the FIPS tests on-demand by the outside world. We expect this to happen in response to a mailbox command while the runtime firmware is running. Unfortunately, the logic for executing the FIPS tests is large (>10k), particularly for the LMS implementation + test vectors, but also for the firmware authentication code. We would prefer to have the option to not include this code in a runtime firmware image.

## Proposal 1: C-ABI callbacks in ROM

[caliptra-sw PR #575](https://github.com/chipsalliance/caliptra-sw/pull/575) adds a C ABI entrypoint to the ROM (uint32_t caliptra_rom_run_fips_tests()) that can be used to execute the FIPS self tests by a future runtime firmware image. I'm also considering exporting another function (uint32 caliptra_rom_validate_mailbox_image()) that can be used to perform the ROM's image validation on the mailbox contents. This may be useful in some hitless update scenarios to make sure the image is compatible with the ROM validation logic before we reset.

Vishal Soni looked at this PR #575, and said there was internal pushback from Microsoft calling back into the ROM. I would like to get more information on what the concerns are, and how they might be mitigated. My position:

1. In Caliptra 1.0, the ROM is always visible to the CPU. Not including these exports doesn't prevent an exploit from jumping back to the ROM, it just provides a standardized, backward-compatible, well-tested way for us to do so from legit firmware.

2. It's likely that we will need to add additional features or CFI mitigations to the runtime firmware after the ROM has frozen, and it might be a challenge to fit everything into ICCM. Having the option to call ROM exports to free up tens of kilobytes could "save the day".

3. The decision about whether to export these functions from the ROM should be made separately from the decision to call them from the runtime firmware. Perhaps the working group would like to say "yes" to adding the exports, but "not yet" to using them in the runtime firmware.

## Proposal 2: When the runtime firmware is asked to re-run the FIPS tests, reset the microcontroller with a special flag to inform the ROM to run the fips tests, validate fmc/runtime, and jump back to FMC.

This is very similar to what we do for hitless updates.

varuns-nvidia, 2023-08-22:
@korran , does the FW expose any FIPS services with LMS? Since the FW doesn't do authentication, I'd think no and thus no FIPS self-tests for LMS are necessary for the FW module. @JohnTraverAmd

korran, 2023-08-22:
Unfortunately, according to @JohnTraverAmd, FIPS requires that we reauthenticate our images, and thus we will need LMS in the runtime.

mhatrevi, 2023-09-18:
The issue needs more discussion and does not meet the v1.0 ROM release timelines. Punting it to v2.0 release.

closed 2023-09-18 by mhatrevi