chipsalliance/caliptra-sw #691: FHT validity is not checked in ROM

nquarton, 2023-08-25:
This was raised as a potential improvement to Vishal and he asked to file as an issue.

The ROM could call is_valid() on the FHT to catch any issues before entering FMC. Currently, any issue will cause a failure in FMC instead.

FerralCoder, 2023-08-25:
If FMC is doing this check at the start anyway, is doubling the check by running it in ROM really adding any value?

nquarton, 2023-08-25:
The idea here would just be to catch an issue earlier in immutable code. Whether or not this specific check actually has a security impact is debatable. It was just something that was noticed where it seemed like ROM checking it's own output would be the best practice.

closed 2023-09-12 by nquarton