chipsalliance/caliptra-sw #695: Don't use eTRNG data verbatim; mix it with previous eTRNG responses and UDS

korran, 2023-08-25:
Data sent to the eTRNG interface may be coming from an SoC microcontroller that has been compromised later in boot. The compromised firmware could feed Caliptra the same bits over and over, convincing the firmware to reuse a nonce, or worse.

varuns-nvidia, 2023-08-25:
Integration requirements require eTRNG to be supplied by a HW gasket with no firmware involvement. It is tougher to audit this though.

korran, 2023-08-25:
From section 8 of Caliptra_Integration_Specification.pdf:

> It is a requirement that this TRNG HW API is always handled by a SOC HW gasket logic (and not some SOC ROM/FW code) for FIPS compliance.

I'm worried that folks that don't care about FIPS will do the easy thing and just implement this in the management CPU.

mhatrevi, 2026-10-26:
Closing this as not applicable to the supported Caliptra 2.2 subsystem configuration.

The [subsystem integration specification](https://github.com/chipsalliance/caliptra-ss/blob/6e5ed764da55bd1ddaf4db7bab0d49666f28e0ab/docs/CaliptraSSIntegrationSpecification.md#L3250) requires CALIPTRA_INTERNAL_TRNG and prohibits use of the external TRNG interface. Consequently, the verbatim external-response path discussed in this issue is not used in that configuration; randomness comes through the internal entropy-source/CSRNG path.

closed 2026-10-26 by mhatrevi