chipsalliance/caliptra-sw #708: Sha384Acc::lock() locks the register with a write instead of a read (as the register docs mention)

korran, 2023-08-30:
https://github.com/chipsalliance/caliptra-sw/blob/8b23c0a6d1fb727c932d2f84b9ca4aa543214da8/drivers/src/sha384acc.rs#L83-L86

This contradicts the [register docs](https://chipsalliance.github.io/caliptra-rtl/main/external-regs/?p=caliptra_top_reg.sha512_acc_csr.LOCK):

> SHA lock register for SHA access, reading 0 will set the lock, Write 1 to clear the lock

korran, 2023-08-30:
Sha384::lock() is only used in fatal error handling, and I believe the intention there is to forcibly steal the lock so the SoC cannot use the peripheral (required for FIPS).

mhatrevi, 2023-09-13:
This is a documentation bug. Filed https://github.com/chipsalliance/caliptra-rtl/issues/216 to update the register doc.

closed 2023-09-16 by mhatrevi