chipsalliance/caliptra-sw #718: Mailbox driver needs to handle error states correctly.

korran, 2023-08-31:
While fixing some driver test cases to pass with the latest RTL, I noticed that there are cases where the SoC can force the mailbox state machine into an error state. The driver does not handle this, and the uC firmware can get stuck after SoC bad behavior (noticed during a uC->SoC transaction, but there are likely other scenarios).

Also, we need better hardware documentation in this area.

calebofearth, 2023-09-01:
@korran The mailbox is only intended to enter an ERROR state if both of these are true:
1. SoC has acquired the lock using a valid PAUSER
2. SoC performs an access outside of the defined order (e.g. write to DLEN prior to write to CMD).

If the uC has lock because it's running a uC->SOC transaction, the mailbox should never be able to enter the ERROR state. If you have a testcase that shows mailbox entering ERROR state while uC has lock, please share it with the HW team and we'll debug.

korran, 2023-09-01:
Yes, that sounds like the scenario I'm hitting. In a uC->SoC transaction, the SoC is clearing the execute bit while the FSM is in the MBOX_EXECUTE_SOC state.

What should the uC be doing when it sees the FSM go into the error state as part of a uC->SoC transaction?  I assume it should eventually write to the unlock register to get things back to normal?

calebofearth, 2023-09-02:
I broke out a new section in the Integration Specification to discuss Mailbox protocol errors and handling. There are two ways to recover from the error - soft reset (by the SOC), or uC writing mbox_unlock to reset the mailbox FSM as you mentioned.

While in the Error state, we haven't prescribed any tasks for the uC to undertake. This state essentially just exists to enforce the protocol rules and not allow SOC agents to interact with the mailbox erroneously. I'm not sure what security related tasks FW may want to perform here - in the future there may be some interest in logging the occurrence of protocol violations to create notifications about misbehaving agents.

rusty1968, 2023-10-02:
@wmaroneAMD

andreslagarcavilla, 2023-11-08:
is this fixed?

mhatrevi, 2023-12-01:
Moving this task to 2.0