chipsalliance/caliptra-sw #726: Get Rand API request

varuns-nvidia, 2023-09-05:
Would be used for supporting a field entropy programming API as requested in https://github.com/chipsalliance/Caliptra/issues/52

jhand2, 2023-09-05:
Ya we can do that. @JohnTraverAmd are there any FIPS considerations is we expose the TRNG to mailbox callers? Since previously we were only using entropy for blinding and DPE handles.

jhand2, 2023-09-11:
@varuns-nvidia to confirm, would it be sufficient to just expose this in RT firmware (and not ROM)?

varuns-nvidia, 2023-09-11:
correct

JohnTraverAmd, 2023-09-13:
Hi @jhand2, this would change our current direction for FIPS.  We had identified that we had no output (including key generation) which required an ESV certified random source.  Therefore, we weren't persuing an ESV certification in 1.0 since there is a significant amount of work related to this.  

I think changing this at this stage of the game is too late.

However, if we enable this as a fuse option, for those who don't care about the FIPS certification or others which are able to get the ESV certification with 1.0, this could be an option.

Could we enable/disable this ability with a fuse?

varuns-nvidia, 2023-09-13:
John, if this is a problem, then I can close this request and change the main spec language to not require field entropy to be provisioned via TRNG.

FerralCoder, 2023-11-01:
can this issue be closed?

varuns-nvidia, 2023-11-01:
Yes, runtime FW is no longer a dependency for field entropy provisioning in the main spec. Closing as not planned.

closed 2023-11-01 by varuns-nvidia