chipsalliance/caliptra-sw #745: C API should support asynchronous messaging

nquarton, 2023-09-08:
The C API should support sending a message and separately polling/completing it. This allows the caller to do other tasks while waiting for Caliptra to respond.

nquarton, 2023-09-08:
This is the proposed change to the API. I plan to make these changes (including any feedback on them) next week.

`
// MAILBOX COMMANDS
// Asyncronous operation:
//  - All commands have an option for asyncronous usage as their last param
//  - Setting this will cause the command function to return once the reqest has been issued to caliptra
//  - The caller should then poll for completion using caliptra_test_for_completion
//  - After successfully polling, the caller should use caliptra_complete to finish the transaction
//    and populate the response buffer originally provided.
//  - The caller MUST ensure the response buffer provided remains available until caliptra_complete returns

// Test for completion of the previously issued mailbox command
// returns: True if the mailbox command is complete
//          False otherwise
bool caliptra_test_for_completion();

// Resets the mailbox and SW state
// Populates the response buffer provided when issueing the command if applicable
// returns: 0 for success
int caliptra_complete();

// Upload Caliptra Firmware
int caliptra_upload_fw(struct caliptra_buffer *fw_buffer, bool async);

// Read Caliptra FIPS Version
int caliptra_get_fips_version(struct caliptra_fips_version *version, bool async);

// Stash Measurement command
int caliptra_stash_measurement(struct caliptra_stash_measurement_req *req, struct caliptra_stash_measurement_resp *resp, bool async);

// Get IDEV CSR
int caliptra_get_idev_csr(struct caliptra_get_idev_csr_resp *resp, bool async);

// Get LDEV Cert
int caliptra_get_ldev_cert(struct caliptra_get_ldev_cert_resp *resp, bool async);

// Execute Mailbox Command
int caliptra_mailbox_execute(uint32_t cmd, struct caliptra_buffer *mbox_tx_buffer, struct caliptra_buffer *mbox_rx_buffer, bool async);
`

bchien, 2023-09-15:
@nquarton can caliptra_complete() be hidden within caliptra_test_for_completion()?

So if caliptra_test_for_completion() returns true, then the caller can safely assume that the response buffer has already been populated (if applicable to the command)

bchien, 2023-09-15:
@nquarton are all the API's listed here available to the caller at all times of the system being ON?  Or some commands are only supported for a finite time window during the Caliptra boot process?

If yes, can we have these marked or called out specifically in the API's?

otherwise, can we assume Caliptra will always return a valid response to each of these calls?  For example, even when command is not supported, Caliptra will return an error indicating so.

nquarton, 2023-09-15:
> @nquarton can caliptra_complete() be hidden within caliptra_test_for_completion()?
> 
> So if caliptra_test_for_completion() returns true, then the caller can safely assume that the response buffer has already been populated (if applicable to the command)

The reason these are separate is because it will take a little bit of time for the copy to happen in complete() whereas the completion check is a quick register read. So in the case where you have a thread handling this communication, for example, you may want to know if the response is ready without actually spending the time to copy it. Then, the thread can be woken up that will handle the completion itself.

nquarton, 2023-09-15:
> @nquarton are all the API's listed here available to the caller at all times of the system being ON? Or some commands are only supported for a finite time window during the Caliptra boot process? [...]

I'm not aware of any restrictions on sending valid commands once runtime firmware is loaded. If you were to send an invalid command, you would get a failure indicating so. This is a non-fatal error and you can send another command after.

That being said, in general, I don't think the intent of the header file would be to go into detail on this kind of documentation. But absolutely feel free to ask me for any clarifications. All the details should be in the caliptra readme files soon. The runtime info is mostly there in https://github.com/chipsalliance/caliptra-sw/blob/main/runtime/README.md. The DPE info is not added yet but I can send you an early version of it (we can't post it on this public github yet).

nquarton, 2023-10-02:
Completed by https://github.com/chipsalliance/caliptra-sw/pull/876

closed 2023-10-02 by nquarton