chipsalliance/caliptra-sw #776: PCR extension inconsistency between FMC and ROM

bluegate010, 2023-09-14:
ROM extends both current and journey PCRs on [cold](https://github.com/chipsalliance/caliptra-sw/blob/9ef6439c93497d75a22c3d48c36675aad431e5a8/rom/dev/src/flow/cold_reset/fw_processor.rs#L121) and [update](https://github.com/chipsalliance/caliptra-sw/blob/9ef6439c93497d75a22c3d48c36675aad431e5a8/rom/dev/src/flow/update_reset.rs#L77) reset. No PCRs are extended on [warm](https://github.com/chipsalliance/caliptra-sw/blob/9ef6439c93497d75a22c3d48c36675aad431e5a8/rom/dev/src/flow/warm_reset.rs#L32) reset.

FMC extends the current PCR on [every kind](https://github.com/chipsalliance/caliptra-sw/blob/9ef6439c93497d75a22c3d48c36675aad431e5a8/fmc/src/flow/rt_alias.rs#L159) of reset, but only extends the journey PCR on cold or update reset.

FerralCoder, 2023-09-14:
This sounds like a mistake in FMC. I assume the current PCR is not being cleared anywhere in the warm reset flow?

mhatrevi, 2023-09-14:
@bluegate010 Can we remove the ROM label from this?

bluegate010, 2023-09-14:
Done. Originally added because I wasn't sure if it was a ROM bug.

FerralCoder, 2023-10-03:
did https://github.com/chipsalliance/caliptra-sw/pull/780 completely address this? Can this be closed?

bluegate010, 2023-10-03:
Yep, closed.