chipsalliance/caliptra-sw #807: Potential DoS waiting for sha384_acc to be locked

korran, 2023-09-15:
Consider this loop, used during image validation on both cold-reset and hitless-update:

https://github.com/chipsalliance/caliptra-sw/blob/9e4f4d673b22a5c23f9d9078872a09c3d02853cb/common/src/verifier.rs#L36-L42

If the SoC sends a hitless-update mailbox command, Caliptra will boot back to the ROM, and the SoC could then lock the sha384_acc peripheral, making the Caliptra ROM freeze up in the above loop, and causing the watchdog to eventually fire.

korran, 2023-09-15:
We might be able to mitigate that one in the runtime firmware by refusing to do the hitless reboot unless we've already locked sha384acc. Need to write a test to ensure the ROM can handle sha384acc already being locked by the runtime.

korran, 2023-09-25:
Note that with the workaround to #841 added in #858, it is no longer possible to fix this issue without a ROM change.

* If we #858 as is, the runtime firmware can't lock the mailbox before resetting to fix https://github.com/chipsalliance/caliptra-sw/issues/807, as that would cause the AssumedLockState::NotAcquried attempt here to timeout in the watchdog.

* If we tweak the #858 to assume ShaAccLockState::Acquired for UpdateReset, we are trusting the runtime firmware too much, as an attacker with some control over the runtime firmware could do an update reset without the lock, and have the SoC grab the lock while the ROM assumes it already has the lock.

mhatrevi, 2023-09-29:
Closing this issue as it is no longer applicable due to https://github.com/chipsalliance/caliptra-sw/pull/862

closed 2023-09-29 by mhatrevi