chipsalliance/caliptra-sw #817: If a warm reset occurs while ICCM is being written during a hitless update, CPU can execute undesired instructions

korran, 2023-09-19:
By controlling the amount of time before the warm reset, an attacker may be able to get the CPU to perform a dangerous operation.

jhand2, 2023-09-19:
One potential fix:

Use the boot status to denote that fw writing is in progress.

On warm reset, first thing ROM does is check the boot status and panic if fw wasn't fully written.  Seems like a decent use-case for using a WarmResetEntry4 data vault slot.

bluegate010, 2023-09-19:
Would another way to put that be, always perform a cryptographic validation of the image before jumping to it, even in a warm-boot flow? Not sure we need to use an extra DV slot for that.

varuns-nvidia, 2023-09-19:
Would the DCCM image pass validation?

korran, 2023-09-19:
We don't necessarily need a datavault slot; We could use an DCCM address in the ROM's reserved  section that we set before we start mutating ICCM and clear after we're done.

korran, 2023-09-19:
| Would another way to put that be, always perform a cryptographic validation of the image before jumping to it,

I think this would be the best option, but I think Bharat or maybe Vishal S wanted the warm reset to occur quickly.

bluegate010, 2023-09-19:
Ah, I misunderstood the suggestion.

korran, 2023-09-19:
| Would the DCCM image pass validation?

I assume you mean ICCM? The mailbox contents would have been validated before the copy to ICCM started, but since the copy was interrupted you would have an attacker-controlled mixture of the new firmware and the old firmware.

Upon warm reset, we do NOT re-validate ICCM, which I think is by design, but I don't recall the reasoning for that.

jhand2, 2023-09-19:
> We don't necessarily need a datavault slot; We could use a DCCM address in the ROM's reserved section that we set before we start mutating ICCM and clear after we're done.

I agree a data vault slot isn't strictly necessary. My only concern was if some other firmware accidentally corrupted that address, then Caliptra would be unrecoverable until next cold reset. Putting it in a lockable DV slot makes that a little more protected.

But I don't feel too strongly either way.

korran, 2023-09-20:
> Putting it in a lockable DV slot makes that a little more protected.

Don't we need to be able to perform N hitless updates? If the DV slot is locked can't it only be used once?

JohnTraverAmd, 2023-09-21:
I would request that we actually verify the FW hash of both the FMC and Run-time and compare it against the locked hash from the DV registers.  FMC TCI is a 48 byte register (Cold Reset Registers) and the RT TCI register (Warm Reset Register).

FIPS requires that the image is verified cryptographically after a reset.

closed 2023-10-10 by mhatrevi