chipsalliance/caliptra-sw #826: Timing Side-Channel When Comparing Context Handles

ericeilertson, 2023-09-20:
Knowledge of a context handle might enable an attacker to impersonate other entities within
the same locality to the DPE command handler. While an attacker could also retrieve context
handles from other localities, it was found these could not be used to execute DPE
commands.

jhand2, 2023-09-20:
Thanks for writing these up! We also have parallel issues for each of these in caliptra-dpe (e.g. https://github.com/chipsalliance/caliptra-dpe/issues/188).

We'll update the labels and copy your descriptions to the issues in caliptra-dpe and then close these ones. Some of the fixes will be in caliptra-sw though, so I need to go through each of these and look at where they should be tracked.

jhand2, 2023-09-29:
Fixed by https://github.com/chipsalliance/caliptra-dpe/pull/199

closed 2023-09-29 by jhand2