chipsalliance/caliptra-sw #827: DestroyCtx Command Can Corrupt Context Tree

ericeilertson, 2023-09-20:
The DestroyCtx command could be used to corrupt the context tree structure, making it
possible for context handles to have incorrect parents or children set, even from different
localities. This flaw could be leveraged by an attacker to call the CertifyKey and Sign
operations on a TCI that has the incorrect parent set, making it attest that it was derived
from an entirely different context handle.

The impact is limited by the fact that the misused entries have to be created after 
DestroyCtx command is called

jhand2, 2023-09-21:
See https://github.com/chipsalliance/caliptra-dpe/issues/189

jhand2, 2023-10-11:
Further fix in https://github.com/chipsalliance/caliptra-dpe/pull/232