chipsalliance/caliptra-sw #828: Multiple Ways to Exhaust Space in DpeInstance::contexts[]

ericeilertson, 2023-09-20:
An attacker that is able to send DPE commands to Caliptra could exhaust the space in the
array that is used to keep track of the context handles. Since the array is shared across
localities, any IP block could make Caliptra unusable for others.

jhand2, 2023-09-20:
See https://github.com/chipsalliance/caliptra-dpe/issues/190 for some more commentary on this. Let's plan to discuss this in one of the firmware syncs since I think an important part of fixing this is settling on the right threat model.

closed 2023-10-04 by jhand2

jhand2, 2023-10-12:
Further fix in https://github.com/chipsalliance/caliptra-sw/pull/944

jhand2, 2023-10-12:
In general, our security model is that we will prevent PL1 from DOSing PL0. Mutual DOS between PL1 PAUSERs is outside the scope of our threat model.