chipsalliance/caliptra-sw #833: Context Handles Not Rotated on Error

ericeilertson, 2023-09-20:
It is assumed that context handle rotation is mandated by the specification to reduce the
risk of commands being replayed in encrypted sessions. Because encrypted sessions were
not in use in the current Caliptra implementation, this risk would not be mitigated in any way
by rotating context handles and the finding is reported for informational purposes only.
Although many types of failures might be expected to reoccur when the same command
was replayed, certain failures caused by hardware glitches could potentially be replayed
successfully at a future time, while other failures due to conditions that are true at one time
but false at another time might also be replayed successfully. An attacker with the ability to
replay such a message might use this to alter the state of the DPE to their advantage.

jhand2, 2023-10-02:
Fixed by https://github.com/chipsalliance/caliptra-dpe/pull/200 and https://github.com/chipsalliance/caliptra-dpe/pull/207

jhand2, 2023-10-02:
Oops, I meant to comment this on #827. Reopening this one.

sree-revoori1, 2023-10-02:
We decided to not do this since rotating a handle on an error would mean that the caller no longer has access to that handle, since if a DPE command fails, we do not return the handle. Since rotating the handle destructively mutates DPE's context array, the old handle is invalid, and the caller has no way to access the new context corresponding to the rotated handle.

Another reason is that rotating the handle on error could be confusing in other ways such as if the input handle itself invalid or if the DPE command failed due to an error in generating the new handle.

closed 2023-10-02 by sree-revoori1

jhand2, 2023-10-12:
Clarification of this behavior in the RT spec: https://github.com/chipsalliance/caliptra-sw/pull/953