chipsalliance/caliptra-sw #835: DeriveChild Permits Context Handle to Coexist With the Default Context

ericeilertson, 2023-09-20:
The implementation does not conform to the specification. The impact of this was not
assessed.
Description
Draft DPE specification says (emphasis by NCC Group):
A DPE SHOULD support default context(s) and may support only default context(s).
If a DPE supports default contexts, it MUST support one default context per
session. A DPE **MUST NOT allow simultaneous use of a default context and
context handles** within the same session: these are mutually exclusive.

However, the implementation does not agree with this. When DPE already has a default
context (this is the state after initialisation) and DeriveChild command is executed with 
retain_parent = true , default = false , none of the following statements will be executed.
This makes is p

jhand2, 2023-09-29:
https://github.com/chipsalliance/caliptra-dpe/issues/197

jhand2, 2023-09-29:
Fixed by https://github.com/chipsalliance/caliptra-dpe/pull/203

closed 2023-09-29 by jhand2

jhand2, 2023-10-12:
Further fixes https://github.com/chipsalliance/caliptra-dpe/pull/235