chipsalliance/caliptra-sw #842: Premature Release of SHA-512 Accelerator

ericeilertson, 2023-09-20:
The SHA-512 Accelerator was a shared component which could be accessed by both the
Caliptra firmware and the SoC, and could be used to compute SHA-384 (and SHA-512)
digests of shared memory regions, in particular the SRAM used to back the Mailbox.
Simultaneous use was restricted by means of a lock.
During the initial startup, Caliptra had exclusive access to this lock in order to perform its
necessary work. By releasing this lock prematurely, this allowed another component to claim
the lock and, by not releasing it, prevent the Caliptra ROM from completing its necessary
validation of the First Mutable Code and Runtime and transferring control to those
components.
The premature release of the lock could also facilitate the exploitation of a more severe
TOCTOU condition described in NCC-MSFT283-YMG, which could result in the ROM loading
unsigned FMC or Runtime code.

JohnTraverAmd, 2023-09-21:
ROM will release the Accel Lock after the KAT has been performed.  Then ROM will acquire the lock when starting the Image verification after a FW load command was received and only release the lock once the verification is completed.

On error, we will not release the lock since our current policy is to go to a fatal error.

mhatrevi, 2023-09-29:
Closing this issue as it is no longer applicable due to https://github.com/chipsalliance/caliptra-sw/pull/862

closed 2023-09-29 by mhatrevi