chipsalliance/caliptra-sw #846: rom_integrity_test Does Not Cover the Currently Empty .data Section

ericeilertson, 2023-09-20:
The ROM boot process includes an integrity test of the ROM itself that does not cover all of
the sections. While the firmware does not currently use .data , if/when it does in future, this
could become an oversight.

On integrity failure the boot process fails and on success the integrity hash is discarded (not
used for any measurements).

korran, 2023-09-20:
There is an assertion in the linker script that the data section is empty:

https://github.com/chipsalliance/caliptra-sw/blob/4bacc8fc47402d331d6528e245d673524a27e738/rom/dev/src/rom.ld#L157-L159

For additional defense in depth, it may be worth moving CALIPTRA_ROM_INFO to its own section after .data.

korran, 2023-10-06:
Fixed by #928 928

closed 2023-10-06 by korran